pkuehnel/TeslaSolarCharger #1301: Cannot change car configuration via http-PUT anymore

BillGOH, 2024-06-08:
I used a script like the following to change charge mode via openHAB:

`curl -X 'PUT' 'http://192.168.202.4:7190/api/Config/UpdateCarConfiguration?carId=1' -H 'accept: */*' -H 'Content-Type: application/json' -d '{ "chargeMode": '$1', "minimumSoC": 30, "latestTimeToReachSoC": "2023-10-29T07:30:00", "ignoreLatestTimeToReachSocDate": true, "maximumAmpere": 16, "minimumAmpere": 4, "usableEnergy": 75, "shouldBeManaged": true, "shouldSetChargeStartTimes": false, "chargingPriority": 2 }'`

However, since latest updates of TSC this does now work. Swagger does not show UpdateCarConfiguration anymore. Any reason for that?

Nevertheless, thanks to @pkuehnel for this great software and his efforts!

pkuehnel, 2024-06-09:
I refactored everything and removed all unused endpoints, Use POST `/api/Index/UpdateCarBaseSettings`. Note: the json you have to post may have changed.

[image]

closed 2024-06-09 by pkuehnel

BillGOH, 2024-06-09:
Thanks for your quick response and help! It works again.

Would love if solution as asked in #840 would be possible.