pkuehnel/TeslaSolarCharger #1326: Two small basic questions

pdb01, 2024-06-17:
Hi again,

Apologies for my maybe irrelevant and stupid questions, but I'm a bit puzzled :-)

Since all the strange Tesla behavior and updates, I also updated some software... anyway, I had to reinstall everything, but since I have these topics:
- I did a back-up via TSC, which is a big file so I expected also the Teslamate database was included. Not sure if it is however, as while I did restore all my previous loggings were gone. First world problems, but was wondering if this is normal?

- I also seem to have lost the "power buffer" visual on the first page. I still use the power buffer to initiate charging (even in case not enough solar power) and it's still working, but not shown on first page neither the used value is reflected on the settings page. Weird or just me? :-)

[image]

pdb01, 2024-06-28:
I try my luck again...

With one of the other questions asked, I saw that indeed (I thought I was getting mad :-p) in the normal view, you should have a power buffer visual, which is gone for me since the latest updates.

[image]
(left is mine; right I took from other question)

Again, functioning is OK as powerbuffer which I impose through a swagger command from Home Assistant is taken into account and functioning, however it's not shown above nor in the settings page (it remains at 0).

As it's functioning, not too big of a concern, but as sometimes reaction is impacted (probably due to the 200 API call limitation etc.), I thought it's worth mentioning.

Regarding my second question (database), not dramatic, but was wondering if indeed TeslaMate full database information is stored in your backup as well. Too late for my past TeslaMate data, but just interesting to understand.

Thanks

pkuehnel, 2024-06-29:
Sorry for not giving an answer. Had the same bug but many things going on currently regarding Tesla API changes. Is fixed in next version

pkuehnel, 2024-06-29:
@pdb01 Please create a separate issue for your other problem

closed 2024-06-29 by pkuehnel