pkuehnel/TeslaSolarCharger #1327: BLE Extension on a seperate Debive (tried rPI 3B)

outrun0506, 2024-06-17:
Hello,

I setup up a fresh Installation of Raspberry Pi OS Lite (64-Bit), installed docker and tried to run the BLE-Extension Container on this seperate Device.

Container starts and is listening. After the first connection I got a response in the Container Log with 404 and the App is Shuting Down.

Any Ideas?

Maybe rPI 3B is not working?

Log output:

```
[17-Jun-2024 14:48:05.033 WRN Microsoft.AspNetCore.Hosting.Diagnostics] Overriding HTTP_PORTS '8080' and HTTPS_PORTS ''. Binding to values defined by URLS instead 'http://+:7210'.
[17-Jun-2024 14:48:05.358 INF Microsoft.Hosting.Lifetime] Now listening on: http://[::]:7210
[17-Jun-2024 14:48:05.360 INF Microsoft.Hosting.Lifetime] Application started. Press Ctrl+C to shut down.
[17-Jun-2024 14:48:05.361 INF Microsoft.Hosting.Lifetime] Hosting environment: Production
[17-Jun-2024 14:48:05.361 INF Microsoft.Hosting.Lifetime] Content root path: /app
[17-Jun-2024 14:50:00.376 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request starting HTTP/1.1 GET http://192.168.0.81:7210/ - null null
[17-Jun-2024 14:50:00.606 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request finished HTTP/1.1 GET http://192.168.0.81:7210/ - 404 0 null 237.8729ms
[17-Jun-2024 14:50:00.632 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request reached the end of the middleware pipeline without being handled by application code. Request path: GET http://192.168.0.81:7210/, Response status code: 404
```

pkuehnel, 2024-06-17:
The Plugin does not host a Page, just insert the value in TSC like described in the Readme.

pkuehnel, 2024-06-17:
Will add a hello World Page https://github.com/pkuehnel/TeslaSolarCharger/issues/1319

closed 2024-06-17 by pkuehnel