pkuehnel/TeslaSolarCharger #1334: TSC finds no BT Beacon

DomiSanNo1, 2024-06-20:
Hey,
TSC BLE API shows following issue:

Error: failed to find BLE beacon for (VIN************* (S6f7b3ba04********): can't scan: context deadline exceeded

Do you know the reason?
BT ist alive, TSC is running on Debian12

pkuehnel, 2024-06-20:
Context deadline exeeded Means the car can't be found within the timout of 20 seconds. As I am only calling Tesla's Command CLI there is nothing I can do about. How far is your Bluetooth device away from the car?

DomiSanNo1, 2024-06-20:
About 10 Meters. The Tesla finds my server when iam searching in the car.

pkuehnel, 2024-06-20:
The infotainment Bluetooth has nothing to do with BLE. 10 Meter is way to much especially when you have walls or other obstacles between the car an your BLE device.

DomiSanNo1, 2024-06-27:
Hey, it works now.
But the API will not be called anymore.

The car is full charged, but TSC Shows up 11kw loading rate.

Any idea?

[image]

pkuehnel, 2024-06-27:
Did the Tesla just finish charging? The actual data is only gathered few minutes (configured in the car settings) as only 200 Requests are allowed per day due to Tesla rate Limits. You can enable using TeslaMate data in the base configuration but very likely TeslaMate won't work anymore in the near future. Tesla claims to phase out old APIs starting April 2024.