pkuehnel/TeslaSolarCharger #1335: Possible to run BLE API on Home Assistant

idominiki, 2024-06-23:
Hi together,

I have a bluetooth dongle connected to HAOS. Is it possible to run BLE API on HAOS, too? So I don't need a second raspberry pi in addition.

Thanks
Dominik

pkuehnel, 2024-06-24:
I have no idea what is needed to run it on HAOS as I never used it. Important is, that your BLE device is near the car.

franciscofsales, 2024-06-24:
One option to integrate with HA is to use the BLE API as a regular API and call it through HA.
That would probably require no changes on the current BLE API, I'd assume.

@pkuehnel are there any docs or info about the BLE API? I tried to look for it and could not find it. Diving into the docker image I can see it is a wrapper around tesla's CLI tool, I'd like to explore it a bit further and potentially work on an integration.
I did look into the TSC ble service and I mean, I can get it from there but I'd like to just know what's going on in the BLE service so that I could make sure I'm not doing something dumb.. is the dockerfile available? I might just be blind but I could not find it.

Thanks

pkuehnel, 2024-06-25:
When you call the BLE API Base URL and add `/swagger` you can find the API docs. TSC does nothing but calling the API and the wrapper does nothing but sending the values to Teslas CLI tool. It results e.g. in:

`/app/go/tesla-control -ble -vin {vin} -key-file {privateKeyLocation} {command} {string.Join(" ", parameters)}`

So basically, it sends the command and all parameters space separated to the CLI tool. Due to potential license issues, I cannot make the BLE API open source.

franciscofsales, 2024-06-25:
> When you call the BLE API Base URL and add `/swagger` you can find the API docs. TSC does nothing but calling the API and the wrapper does nothing but sending the values to Teslas CLI tool. It [...]

Thanks for the info.
Would you be willing to share the dockefile? That should not have any licensing limitations, right?

Either way, that is enough for me to work with.

pkuehnel, 2024-06-26:
@franciscofsales
```
# Stage 1: Build the Go application
FROM golang:1.22 as go
WORKDIR /app

RUN apt-get update \
    && apt-get install -y curl

RUN curl -L https://api.github.com/repos/teslamotors/vehicle-command/tarball/main | tar -xz --strip=1
RUN go get ./...
RUN go build ./...
RUN go install ./...

# Stage 2: Build the .NET application
FROM mcr.microsoft.com/dotnet/sdk:8.0-bookworm-slim-amd64 AS build
ARG BUILD_CONFIGURATION=Release
WORKDIR /src
COPY ["TeslaSolarCharger.BleApi/TeslaSolarCharger.BleApi.csproj", "TeslaSolarCharger.BleApi/"]
RUN dotnet restore "./TeslaSolarCharger.BleApi/TeslaSolarCharger.BleApi.csproj"
COPY . .
WORKDIR "/src/TeslaSolarCharger.BleApi"
RUN dotnet build "./TeslaSolarCharger.BleApi.csproj" -c $BUILD_CONFIGURATION -o /app/build

# Stage 3: Publish the .NET application
FROM build AS publish
ARG BUILD_CONFIGURATION=Release
RUN dotnet publish "./TeslaSolarCharger.BleApi.csproj" -c $BUILD_CONFIGURATION -o /app/publish /p:UseAppHost=false

# Stage 4: Final image
FROM mcr.microsoft.com/dotnet/aspnet:8.0 AS final
WORKDIR /app
EXPOSE 80
ENV ASPNETCORE_URLS=http://+:80

# Copy Go application
COPY --from=go /go/bin /app/go

# Copy .NET application
COPY --from=publish /app/publish .

ENTRYPOINT ["dotnet", "TeslaSolarCharger.BleApi.dll"]

```