pkuehnel/TeslaSolarCharger #1337: Indication if BLE device is not reachable or sending command failed

Dirk77777, 2024-06-25:
I occasionally have the issue that the device running the container with the BLE proxy is not reachable.

It would be nice if TSC could display an indication on the main screen if sending a BLE command failed.

pkuehnel, 2024-06-25:
Many things planned in that front. Update was just a Quick and dirty solution so it works if everything is perfect. For now there does not exist any error handling or such things.

pkuehnel, 2024-10-06:
From the next release, TSC will autodetect if a BLE Command fails and fallback to Fleet API. TSC can also detect the installed BLE Container version, so reminds you if you need to update the BLE container due to API changes.

closed 2024-10-06 by pkuehnel