pkuehnel/TeslaSolarCharger #1343: BLE error message

AlexanderTreptow, 2024-06-30:
Hi,

I get the blue box when clicking to connect and the red error message when trying to test the connection:

[image]

pkuehnel, 2024-06-30:
Look into BLE container logs after clicking on pair. Something is wrong when you try to pair.

AlexanderTreptow, 2024-06-30:
Here are the logs:

```
[30-Jun-2024 17:26:32.712 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request finished HTTP/1.1 GET http://raspberrypi:7210/api/Pairing/PairCar?vin=LRWxxx - 500 0 null 20.7259ms
[30-Jun-2024 17:26:38.232 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request starting HTTP/1.1 POST http://raspberrypi:7210/api/Command/ExecuteCommand?vin=LRWxxx&command=charging-set-amps - application/json; charset=utf-8 null
[30-Jun-2024 17:26:38.233 INF Microsoft.AspNetCore.Routing.EndpointMiddleware] Executing endpoint 'TeslaSolarCharger.BleApi.Controllers.CommandController.ExecuteCommand (TeslaSolarCharger.BleApi)'
[30-Jun-2024 17:26:38.233 INF Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker] Route matched with {action = "ExecuteCommand", controller = "Command"}. Executing controller action with signature System.Threading.Tasks.Task`1[TeslaSolarCharger.BleApi.Dtos.DtoCommandResult] ExecuteCommand(System.String, System.String, System.Collections.Generic.List`1[System.String]) on controller TeslaSolarCharger.BleApi.Controllers.CommandController (TeslaSolarCharger.BleApi).
[30-Jun-2024 17:26:38.236 VRB TeslaSolarCharger.BleApi.Services.CommandService] ExecuteCommand(LRWxxx, charging-set-amps, ["7"]
[30-Jun-2024 17:26:38.236 VRB TeslaSolarCharger.BleApi.Services.CommandService] Waiting for semaphoreSlim to allow command execution in a9d5772f-fce8-4b95-9d09-24afe7d8865a
[30-Jun-2024 17:26:38.236 VRB TeslaSolarCharger.BleApi.Services.CommandService] SemaphoreSlim allowed command execution
[30-Jun-2024 17:26:38.237 VRB TeslaSolarCharger.BleApi.Services.CommandLineExecutionService] ExecuteCommand(/app/go/tesla-control, -ble -vin LRWxxx -key-file /externalFiles/privateKey.pem charging-set-amps 7)
[30-Jun-2024 17:26:38.239 VRB TeslaSolarCharger.BleApi.Services.CommandLineExecutionService] Using execution timout of 21 seconds
[30-Jun-2024 17:26:38.257 VRB TeslaSolarCharger.BleApi.Services.CommandLineExecutionService] Process exited: True
[30-Jun-2024 17:26:38.258 VRB TeslaSolarCharger.BleApi.Services.CommandLineExecutionService] Stdout Result: 
[30-Jun-2024 17:26:38.258 VRB TeslaSolarCharger.BleApi.Services.CommandLineExecutionService] Process completed with exit code 1
[30-Jun-2024 17:26:38.258 VRB TeslaSolarCharger.BleApi.Services.CommandLineExecutionService] Stderr Result: Error loading credentials: open /externalFiles/privateKey.pem: no such file or directory

[30-Jun-2024 17:26:38.259 VRB TeslaSolarCharger.BleApi.Services.CommandService] Waiting 1000 ms before allowing next command execution
[30-Jun-2024 17:26:38.259 INF Microsoft.AspNetCore.Mvc.Infrastructure.ObjectResultExecutor] Executing ObjectResult, writing value of type 'TeslaSolarCharger.BleApi.Dtos.DtoCommandResult'.
[30-Jun-2024 17:26:38.260 INF Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker] Executed action TeslaSolarCharger.BleApi.Controllers.CommandController.ExecuteCommand (TeslaSolarCharger.BleApi) in 26.2016ms
[30-Jun-2024 17:26:38.260 INF Microsoft.AspNetCore.Routing.EndpointMiddleware] Executed endpoint 'TeslaSolarCharger.BleApi.Controllers.CommandController.ExecuteCommand (TeslaSolarCharger.BleApi)'
[30-Jun-2024 17:26:38.261 INF Microsoft.AspNetCore.Hosting.Diagnostics] Request finished HTTP/1.1 POST http://raspberrypi:7210/api/Command/ExecuteCommand?vin=LRWxxx&command=charging-set-amps - 200 null application/json; charset=utf-8 28.8602ms
```
Thanks in advance.

pkuehnel, 2024-06-30:
The interesting part is before the first line. The log starts when ` Request finished HTTP/1.1 GET http://raspberrypi:7210/api/Pairing/PairCar?vin=LRWxxx - 500 0 null 20.7259ms` is finished.

AlexanderTreptow, 2024-06-30:
Here you go :)

`
[image]
`

pkuehnel, 2024-06-30:
How does your docker-compose.yml BLE part look like?

AlexanderTreptow, 2024-06-30:
Here is my .yml:

```
version: '3.3'

services:
  teslamate:
    image: teslamate/teslamate:latest
    restart: always
    environment:
      - DATABASE_USER=teslamate
      - DATABASE_PASS=secret ##You can change your password here
      - DATABASE_NAME=teslamate
      - DATABASE_HOST=database
      - MQTT_HOST=mosquitto
      - ENCRYPTION_KEY=supersecret ##You can change your encryption key here
      - TZ=Europe/Berlin ##You can change your Timezone here
    ports:
      - 4000:4000
    volumes:
      - ./import:/opt/app/import
    cap_drop:
      - all

  database:
    image: postgres:15
    restart: always
    environment:
      - POSTGRES_USER=teslamate
      - POSTGRES_PASSWORD=secret ##You can change your password here
      - POSTGRES_DB=teslamate
    volumes:
      - teslamate-db:/var/lib/postgresql/data

  grafana:
    image: teslamate/grafana:latest
    restart: always
    environment:
      - DATABASE_USER=teslamate
      - DATABASE_PASS=secret ##You can change your password here
      - DATABASE_NAME=teslamate
      - DATABASE_HOST=database
    ports:
      - 3100:3000
    volumes:
      - teslamate-grafana-data:/var/lib/grafana

  mosquitto:
    image: eclipse-mosquitto:2
    restart: always
    command: mosquitto -c /mosquitto-no-auth.conf
    #ports:
    #  - 1883:1883
    volumes:
      - mosquitto-conf:/mosquitto/config
      - mosquitto-data:/mosquitto/data

  teslamateapi:
    image: tobiasehlert/teslamateapi:latest
    logging:
        driver: "json-file"
        options:
            max-file: "5"
            max-size: "10m"
    restart: always
    depends_on:
      - database
    environment:
      - DATABASE_USER=teslamate
      - DATABASE_PASS=secret ##You can change your password here
      - DATABASE_NAME=teslamate
      - DATABASE_HOST=database
      - MQTT_HOST=mosquitto
      - TZ=Europe/Berlin ##You can change your Timezone here
      - ENABLE_COMMANDS=true
      - COMMANDS_ALL=true
      - API_TOKEN_DISABLE=true
      - ENCRYPTION_KEY=supersecret ##You can change your encryption key here
    #ports:
    #  - 8080:8080

  teslasolarcharger:
    image: pkuehnel/teslasolarcharger:latest
    container_name: teslasolarcharger
    logging:
        driver: "json-file"
        options:
            max-file: "10"
            max-size: "100m"
    restart: always
    depends_on:
      - teslamateapi
    environment:
#      - Serilog__MinimumLevel__Default=Verbose #uncomment this line and recreate container with docker compose up -d for more detailed logs
      - TZ=Europe/Berlin ##You can change your Timezone here
    ports:
      - 7190:80
    volumes:
      - teslasolarcharger-configs:/app/configs
  
  modbusplugin:
    image: pkuehnel/teslasolarchargermodbusplugin:latest
    container_name: teslasolarcharger_modbusplugin
    logging:
        driver: "json-file"
        options:
            max-file: "5"
            max-size: "10m"
    restart: always
    environment:
      - RequestBlockMilliseconds=0
    ports:
      - 7191:80
      
  bleapi:
    image: ghcr.io/pkuehnel/teslasolarchargerbleapi:latest
    container_name: TeslaSolarChargerBleApi
    privileged: true
    restart: unless-stopped
    network_mode: host
    environment:
      - ASPNETCORE_URLS=http://+:7210

volumes:
  teslamate-db:
  teslamate-grafana-data:
  mosquitto-conf:
  mosquitto-data:
  teslasolarcharger-configs:
  tscbleapi:
```

pkuehnel, 2024-06-30:
after environment the volumes are missing

```
services:
#here are all the other services like TeslaMate, TSC, etc.
  bleapi:
    image: ghcr.io/pkuehnel/teslasolarchargerbleapi:latest
    container_name: TeslaSolarChargerBleApi
    privileged: true
    restart: unless-stopped
    network_mode: host
    environment:
      - ASPNETCORE_URLS=http://+:7210
    volumes:
      - tscbleapi:/externalFiles
      - /var/run/dbus:/var/run/dbus

volumes:
  #here are all the other volumes like teslamate-db, teslamate-grafana-data, etc.
  tscbleapi: 
```

AlexanderTreptow, 2024-06-30:
It's working perfectly now. Thank you very much for your support and creating this wonderful app :).

closed 2024-06-30 by pkuehnel