pkuehnel/TeslaSolarCharger #1344: Switch to control the charging station for charging currents above 5A, set Tesla to 16A

ptar, 2024-06-30:
A possible solution to get around the new Tesla API's request count limit might be to switch to station control for charge currents of 6A and above, and keep Tesla control for currents below 6A.

Therefore, a HTTP, REST or similar command to set the charging amps of the charging station must be available and must be provided in the TSC settings.
* See e.g. https://github.com/goecharger/go-eCharger-API-v2/blob/main/http-en.md for examples 

**Desired charge current >=6A**
If the charge current should be set to 6A or higher, e.g. 8A
* Set the maximum **charge current of the charging station** to the **desired charge current** (8A).
* If the maximum charge current of the **Tesla** is less than 16A, set it to **16A**.

Then check the charging current in the charging station and leave the Tesla at 16A until the desired charging current drops below 6A.

**Desired charge current <6A**
If the desired charge current drops below 6A, e.g. 5A:

* Set **Tesla charge current** to **desired charge current** (5A)
    * This is the current behavior of the TSC for all current settings.
 * If the maximum charge current of the **charging station** is below 16A, set it to **16A**.

Thank you for considering this suggestion!

pkuehnel, 2024-06-30:
That already came to my mind. But will take months to implement. Primary goal is to have a solution that works for everyone. Secondary goal is to implement OCPP. I won't implement custom APIs.

pkuehnel, 2024-06-30:
The generic way for now is to optimize BLE. Tesla reduced the limit for charging commands to 5 per car per day, so it makes the API completely useless for TSC.

ptar, 2024-06-30:
I agree, BLE is the right way with the command limitations in mind.
Nevertheless, if the general logic for using a different device for >= 6A and < 6A is implemented, it wouldn't matter, if a single URL is called or OCPP is used. That's just like getting the grid power via a pre-configured URL.