pkuehnel/TeslaSolarCharger #1399: Define a clean exit to the source configuration window

ptar, 2024-07-18:
The Source configuration window may only be exited when you select Save or Cancel.

Currently, any (accidental) mouse click outside of the window closes the window immediately, resulting in the loss of the currently edited configuration.

Combine this with https://github.com/pkuehnel/TeslaSolarCharger/issues/1398 and you would have the perfect pain in the ass.

pkuehnel, 2024-07-20:
So
1. Clicking outside the dialog should not close it
2. There should be an x to close the dialog instead of only being able to close by clicking on `cancel` or `save`

Correct?

ptar, 2024-07-20:
1. Correct
2. IMHO: X and cancel should do the same: 
  * If **no** values are changed: Close the window
  * If values are changed, the user should be asked:
    * "Do you want to close the window and discard the changes?"
      * "Yes" / "OK" => Close and discard the changes
      * "No" / "Cancel" => Stay in the window keeping the current changes
      * "Save and close" => Save changes and close window

pkuehnel, 2024-09-29:
Backdrop click and escape button disabled in Solar value configuration dialogs

closed 2024-09-29 by pkuehnel