pkuehnel/TeslaSolarCharger #1404: BLE API issues with

Lord225, 2024-07-22:
Hello,
I'm unable to get bleapi to work properly.
Hardware: Raspberry Pi 4 (B)
I tried configuring it with the docker-compose provided in the README, but it throws a "connection refused" error. I had to create a network and set the driver to host. Now, I am able to reach the service.
```
  bleapi:
    image: ghcr.io/pkuehnel/teslasolarchargerbleapi:latest
    container_name: TeslaSolarChargerBleApi
    privileged: true
    restart: unless-stopped
    networks:
      - network
    environment:
      - ASPNETCORE_URLS=http://+:7210
    volumes:
      - /home/lord225/tscbleapi:/externalFiles
      - /var/run/dbus:/var/run/dbus
networks:
  network:
    driver: host
      ```
but unfortunately, it still displays an error message about hci not being able to crate socket. 
      ```
      Error: failed to find a BLE device: can't init hci: can't create socket: address family not supported by protocol
      ```
[ble-logs.txt](https://github.com/user-attachments/files/16329919/ble-logs.txt)
```
lord225@raspberrypi:~ $ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; preset: enabled)
     Active: active (running) since Sun 2024-07-21 17:14:13 CEST; 16h ago
       Docs: man:bluetoothd(8)
   Main PID: 5563 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 1576)
        CPU: 934ms
     CGroup: /system.slice/bluetooth.service
             └─5563 /usr/libexec/bluetooth/bluetoothd
```
```
lord225@raspberrypi:~ $ ls /var/run/dbus
containers  system_bus_socket
```

pkuehnel, 2024-07-24:
Sorry I can not help you debugging the issue. I am just using Teslas Command CLI tool and have no idea what is happening inside of it. What I can tell: On a clean install of Raspbian OS on a Raspberry Pi 4 it is working.

Lord225, 2024-07-25:
If you find this suggestion worthy, I might create a pull request. However, since it involves only about 10 lines of code, it's not really worth the effort for me to set up the environment.

pkuehnel, 2024-07-25:
> I would like to ask for exac Raspbian version bsc you can install like 3 diffrent ones from Raspberry Imager alone

I used the default Raspberry Pi OS 64 bit from the official Raspberry Pi Imager.

> However, I found a new issue and am confused about how to approach it. My colleague is reporting the following error:
> 
> ```
> Error: failed to find BLE beacon for XP7YGCEKXPB214752 (S7f957137799819efC): can't scan: context deadline exceeded
> ```

The error should come up if the Raspberry cannot find the car, e.g. because it is too far away. I am not sure why, but some told me it works about 15 meters away from the car. My experience is that everything further than 4–5 meters away or any wall between Raspberry and car leads to issues.

> I found out the hard way that TeslaSolarCharger will not back off BLE requests if they fail one after another, causing it to spam with requests. I suggest adding a feature in private async Task<DtoGenericTeslaResponse?> SendCommandToTeslaApi in TeslaFleetApiService.cs [code](https://vscode.dev/github/pkuehnel/TeslaSolarCharger/blob/master/TeslaSolarCharger/Server/Services/TeslaFleetApiService.cs#L491) to prevent BLE requests if the previous one failed.

The BLE implementation was just a quick and super dirty fix regarding the rate limits. There is no error handling at all, and the user does not get any information about the errors. As soon as I have time, I will resolve these issues.

pkuehnel, 2024-09-28:
On BLE error TSC will auto fallback to Fleet API on non success BLE calls. After a non success BLE call, TSC will use Fleet API for 5 minutes until BLE is tried another time. Configurable via Environment variable `- BleUsageStopAfterErrorSeconds=300`

closed 2024-09-28 by pkuehnel

Lord225, 2024-09-28:
Here’s the grammar fix for your text:

I see that you're mentioning this issue elsewhere. Here's my workaround that is currently working for me:

If the container is throwing this error, you should check whether the same error occurs when you run the vehicle-command natively on a Raspberry Pi.

To do this, install go on the Raspberry Pi, and then build and install [vehicle-command](https://github.com/teslamotors/vehicle-command) on the Raspberry Pi (you'll need to clone the repository and run go install as stated in the instructions). If you do this correctly, you will be able to use tesla-control natively on the Raspberry Pi.
```
// Generate a pair of encryption keys
openssl ec -in private.pem -pubout > public.pem
// Ask the car to add a new key
tesla-control -debug -ble -vin XXXXXXXX add-key-request ./public.pem owner cloud_key
```
```
// Check if it is working (honk the car)
tesla-control -debug -domain vcsec -ble -vin XXXXXXXX -key-file ./private.pem honk
```
If it works, you can blame Docker or your Raspbian installation for not functioning correctly with BLE.

or you can run python script that will mockup original container what is not ideal but works...
```py
import os
import logging
from flask import Flask, request, jsonify

logger = logging.getLogger('ble-api')

app = Flask(__name__)

BASE_PATH = "/home/lord225/tscbleapi"
KEY_PATH = os.path.join(BASE_PATH, "private.pem")
SESSION_CACHE = os.path.join(BASE_PATH, "cache.txt")

def run_command(vin, command):
    """Executes the command for the given VIN."""
    cmd = (
        f"/home/lord225/go/bin/tesla-control -debug -ble -session-cache "
        f"{SESSION_CACHE} -vin {vin} -key-file {KEY_PATH} {command}"
    )
    
    logger.info(f"Executing command: {cmd}")
    exit_status = os.system(cmd)
    
    if exit_status == 0:
        logger.info(f"Command {command} executed successfully")
        return True
    else:
        logger.error(f"Command {command} failed with exit status {exit_status}")
        return False

def create_response(status_code, message, success):
    """Helper function to create a standardized JSON response."""
    return jsonify({
        "StatusCode": status_code,
        "Message": message,
        "Success": success
    }), status_code

@app.route("/api/Ble/honk", methods=['POST'])
def honk():
    vin = request.args.get('vin')

    if not vin:
        return create_response(400, "Missing 'vin' parameter.", False)

    success = run_command(vin, "honk")
    return create_response(200, "Success" if success else "Failed", success)

@app.route("/api/Command/ExecuteCommand", methods=['POST'])
def execute_command():
    data = request.get_json()
    vin = request.args.get('vin')
    command = request.args.get('command')

    if not vin:
        return create_response(400, "Missing 'vin' parameter.", False)

    if not command:
        return create_response(400, "Missing 'command' parameter.", False)

    # Concatenate all values from the data array into a single string
    command_params = ' '.join(map(str, data))
    full_command = f"{command} {command_params}"

    success = run_command(vin, full_command)
    return create_response(200, "Success" if success else "Failed", success)

if __name__ == "__main__":
    app.run(host='0.0.0.0', port=7210)
````

pkuehnel, 2024-09-28:
As I do not have any issues on two different locations I can not test it properly. But thanks for your input. Just for your info: The TSC BLE Container does nothing else than using Tesla's Go library and return the CLI result via the REST API.

pkuehnel, 2024-09-28:
Just another info: In the new Update the REST API interface will change.