pkuehnel/TeslaSolarCharger #1406: Min SOC + PV doesn't work as described

illuzn, 2024-07-26:
Use the following json to fake solar power generation numbers:

http://echo.jsontest.com/grid/0/battery/0/solar/10000

Despite there being 10kW of solar my tesla refuses to charge above the minimum SoC even after the time out period has run.

Setting the minimum SoC starts charging immediately on grid power so that it is not an API issue.

pkuehnel, 2024-07-26:
It is because grid is used as source to determine how fast the car should charge. The goal is to not feed power to the grid and not consume Power from the grid.

illuzn, 2024-07-27:
So if grid is +10000W it will charge? I notice that negative numbers don't work (assumes grid is always the grid export).

pkuehnel, 2024-07-27:
Negative numbers do work. If grid is +5000 Watt charging power will be increased by 5000 Watt, if it is -5000 Watt, charging power will be decreased by 5000 Watt.

illuzn, 2024-07-27:
Okay... now I understand, will post a fix to the readme to clarify the plus/ minus correction factor.

I thought they literally performed addition/ subtraction of a fixed number (hence I was deleting the 1.0 that is there by default and changing it to 0). This resulted in zeros for the readings no matter what I was doing.

Doh!

closed 2024-07-27 by illuzn