pkuehnel/TeslaSolarCharger #1420: After Update to 2024.20.6 SetChargingAmps failed :edge

b1ggi, 2024-08-12:
Hi, since the car update BLE cannot set charging amps. 

Failed to execute command: car could not execute command: SetChargingAmps failed

bleproxy host restart, repairing does not work. 

I am running :edge tsc, i tried :latest and :edge bleproxy

pkuehnel, 2024-08-13:
Very likely a bug in the car's firmware. The charging current can not be Set via the Tesla API as well, correct?

b1ggi, 2024-08-13:
I can test this evening

b1ggi, 2024-08-14:
Yes same with ble disabled, only api (156 set amp api calls)

pkuehnel, 2024-08-14:
Sorry I meant the Tesla App, not API. The Official Tesla App also can not change current.

pkuehnel, 2024-08-15:
Issue is fixed with Tesla firmware 2024.26.7

pkuehnel, 2024-08-16:
Looks like the update does not solve the issue for everyone.

b1ggi, 2024-08-20:
.8 solved it for me just now

pkuehnel, 2024-08-21:
Some others also claimed that after an update from .6 to .7 solved the issue, so looks like an update solves the problem. Unfortunately, a software reinstall from the service menu does not help (at least not for me).

DerHexer, 2024-08-22:
This morning I started charging (combined with anprobably unrelated reboot of my BLE RPi).
As everything looked good last evening, it currently (for me) is fixed to command my Tesla via TSC as well.
The test-setting to 7A works, as well as automatic solar-based charging.
As mentioned above, version .7 introduced it and one reboot of my M3 2021 seems to have fixed it. YMMV, but maybe not all people with this issue tried a reboot yet and for some it helps too 🤷‍♂️

pkuehnel, 2024-08-23:
Looks like I found a solution. What you need to do: Disconnect the car charger, sit in the car, switch off the Bluetooth of your phone. Make sure that no BLE enabled phone, Key Fob,... is around. Note: BLE has a connection radius of > 20 meters, so best would be to disable Bluetooth of all phones and remove the battery from the key fob. Then restart the car by pressing both scroll wheels, wait for the car to reboot (I waited an additional minute after the navigation screen came back), enable Bluetooth on your device. Now it should work again.

b1ggi, 2024-08-23:
Nice find, unfortunately i cant test/verify it anymore, may be useful for future maybe

DerHexer, 2024-08-27:
Well, it‘s me again. I had to reboot my Tesla due to a black screen yesterday evening.
Unfortunately today I noticed that my previous fix no longer seems to be working. I am back to being unable to set amps via app or TSC.
Followed the recommended steps by pkuehnel multiple times. Powered of the BLE Rpi, turned off iPhone BT. Unfortunately it’s not working for me.

b1ggi, 2024-09-12:
I had the bug again after update to 2024.26.8. rebooting without ble in range did not help. I reported the bug to Tesla in the App, they called me and asked about it. They acknowledged its a bug on their side, no estimation when fixed. Then suddenly the bug disappeared without updating or doing something specific.

pkuehnel, 2024-09-24:
Fixed in Tesla Firmware 2024.32.3

closed 2024-09-24 by pkuehnel