pkuehnel/TeslaSolarCharger #1452: Usability improvement: Add space below Submit-Button on Base-Configuration page

ralphsiegert, 2024-09-01:
On my IPhone the submit button is very close to the lower end of the display and sometimes it takes 2-3 attempts to press it properly that it saves the values. The task switcher from IOS is also very close, so I think this could also interfere with the button and my fingers. 
I suggest to add a padding below or just 2 <BR> below the button. 

[image]

Additionally, I think it would be good to lock the button after I hit it or to give visual feedback that I hit it (this looks like a different story, I add this separately if you want to).

pkuehnel, 2024-09-01:
Yes, please add another issue vor adding visual feedback, thanks.

pkuehnel, 2024-09-08:
Will be fixed in the next version

closed 2024-09-08 by pkuehnel