pkuehnel/TeslaSolarCharger #1467: Grid Power isn't avaivable

DomiSanNo1, 2024-09-11:
Grid Power doenst get states. I've chenged nothing in the config.

[image]
[image]

pkuehnel, 2024-09-12:
Looks like you are using the SMA Plugin. As SMA sends the power data via multicast very likely a network change is responsible for that:
 * Do you have a new switch in the network?
 * Do you have a new router in the network?
 * Did you change any thing with the cabling?
 * Did you switch your OS from Linux to Windows? On Windows the SMA plugin is not supported

pkuehnel, 2024-09-12:
I am pretty sure it is a network problem. If you cannot fix it, you can define fix IP Adresses in the SunnyPortal, where the messages should go to. Note: You need to add all your inverters and your device with the SMA plugin.

DomiSanNo1, 2024-09-12:
I've filled the unicast fields at the sunnyportal (inverter + TSC Server). What do i have to configure in TSC? I get no values.

pkuehnel, 2024-09-12:
Nothing, it should just work. What does the SMA Plugin Log say?

DomiSanNo1, 2024-09-12:
How can I read it?

pkuehnel, 2024-09-12:
Is the timestamp of the latest logs correct? Might be in UTC Time. Is there any whitespace character in the URL because TSC says `Requesting string with url http://192.168.1.200:7192/api/CurrentPower/GetPower?`. The last `?` should not be there.
What happens when you open `http://192.168.1.200:7192/swagger` in your web browser? Are you able to get the value there?
[image]

DomiSanNo1, 2024-09-12:
No I dont 
[image]

pkuehnel, 2024-09-12:
Then your sma Plugin still does not receive the Power values. There is an issue with your Network. When it is working you should See a new Log entry with the Power values in the SMA Plugin at least once every second.

pkuehnel, 2024-09-12:
Can you send me a logfile of the SMA Plugin here?

DomiSanNo1, 2024-09-14:
I can collect the values via HomeAssistant and save them in my MQTT. TSC can read the MQTT data and it works for now. But TSC don't find the HomeManager directly. I reached my personal limit ;)

DomiSanNo1, 2024-09-15:
The containers in docker looking strange too. What does the "-1" mean?

[image]

pkuehnel, 2024-09-15:
Normal naming behavior of docker `directory-imageName-number`. The TSC Containers have fixed names defined in the `docker-compose.yml` as some parts of the Readme (e.g. Generate logs) depend on the Container names.

pkuehnel, 2024-09-15:
So dokumente because your yml is in the Direktory `Dokumente` and -1 because one Container is running. If you would have a second container with the same Service Name it would be -2

closed 2024-09-24 by pkuehnel