pkuehnel/TeslaSolarCharger #1562: after update to new release latest: no such column: c.RateLimitedUntil

b1ggi, 2024-10-16:
Hello,

i can not switch from :edge to :latest

docker logs:
`2024-10-16T21:06:08.954902621Z [16-Oct-2024 23:06:08.954 WRN Microsoft.AspNetCore.DataProtection.Repositories.FileSystemXmlRepository] Storing keys in a directory '/root/.aspnet/DataProtection-Keys' that may not be persisted outside of the container. Protected data will be unavailable when container is destroyed. For more information go to https://aka.ms/aspnet/dataprotectionwarning
2024-10-16T21:06:09.033601795Z [16-Oct-2024 23:06:09.032 WRN Microsoft.AspNetCore.DataProtection.KeyManagement.XmlKeyManager] No XML encryptor configured. Key xyz may be persisted to storage in unencrypted form.
2024-10-16T21:06:09.040971018Z [16-Oct-2024 23:06:09.040 WRN Microsoft.AspNetCore.Hosting.Diagnostics] Overriding HTTP_PORTS '8080' and HTTPS_PORTS ''. Binding to values defined by URLS instead 'http://+:80'.
2024-10-16T21:06:09.248837605Z [16-Oct-2024 23:06:09.248 INF TeslaSolarCharger.Server.Services.CoreService] Backup for this version already created. No new backup needed.
2024-10-16T21:06:09.261884082Z [16-Oct-2024 23:06:09.261 INF TeslaSolarCharger.Server.Services.TelegramService] Can not send Telegram Message because botkey is empty.
2024-10-16T21:06:09.345190852Z [16-Oct-2024 23:06:09.345 DBG TeslaSolarCharger.Server.Services.ConfigJsonService] Use 233V for charge speed calculation
2024-10-16T21:06:09.470659346Z [16-Oct-2024 23:06:09.470 ERR Microsoft.EntityFrameworkCore.Database.Command] Failed executing DbCommand (2ms) [Parameters=[], CommandType='Text', CommandTimeout='30']
2024-10-16T21:06:09.470670527Z SELECT "c"."Id", "c"."ChargeMode", "c"."ChargerActualCurrent", "c"."ChargerPhases", "c"."ChargerPilotCurrent", "c"."ChargerRequestedCurrent", "c"."ChargerVoltage", "c"."ChargingPriority", "c"."ClimateOn", "c"."IgnoreLatestTimeToReachSocDate", "c"."LatestTimeToReachSoC", "c"."Latitude", "c"."Longitude", "c"."MaximumAmpere", "c"."MinimumAmpere", "c"."MinimumSoc", "c"."Name", "c"."PluggedIn", "c"."RateLimitedUntil", "c"."ShouldBeManaged", "c"."ShouldSetChargeStartTimes", "c"."SoC", "c"."SocLimit", "c"."State", "c"."TeslaFleetApiState", "c"."TeslaMateCarId", "c"."UsableEnergy", "c"."UseBle", "c"."VehicleCommandProtocolRequired", "c"."Vin"
2024-10-16T21:06:09.470676368Z FROM "Cars" AS "c"
2024-10-16T21:06:09.479207120Z [16-Oct-2024 23:06:09.478 ERR Microsoft.EntityFrameworkCore.Query] An exception occurred while iterating over the results of a query for context type 'TeslaSolarCharger.Model.EntityFramework.TeslaSolarChargerContext'.
2024-10-16T21:06:09.479217800Z Microsoft.Data.Sqlite.SqliteException (0x80004005): SQLite Error 1: 'no such column: c.RateLimitedUntil'.
2024-10-16T21:06:09.479221177Z    at Microsoft.Data.Sqlite.SqliteException.ThrowExceptionForRC(Int32 rc, sqlite3 db)
2024-10-16T21:06:09.479236014Z    at Microsoft.Data.Sqlite.SqliteCommand.PrepareAndEnumerateStatements()+MoveNext()
2024-10-16T21:06:09.479238619Z    at Microsoft.Data.Sqlite.SqliteCommand.GetStatements()+MoveNext()
2024-10-16T21:06:09.479241525Z    at Microsoft.Data.Sqlite.SqliteDataReader.NextResult()
2024-10-16T21:06:09.479244390Z    at Microsoft.Data.Sqlite.SqliteCommand.ExecuteReader(CommandBehavior behavior)
2024-10-16T21:06:09.479246915Z    at Microsoft.Data.Sqlite.SqliteCommand.ExecuteReaderAsync(CommandBehavior behavior, CancellationToken cancellationToken)
2024-10-16T21:06:09.479249770Z    at Microsoft.Data.Sqlite.SqliteCommand.ExecuteDbDataReaderAsync(CommandBehavior behavior, CancellationToken cancellationToken)
2024-10-16T21:06:09.479252595Z    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteReaderAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
2024-10-16T21:06:09.479255270Z    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteReaderAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
2024-10-16T21:06:09.479257825Z    at Microsoft.EntityFrameworkCore.Query.Internal.SingleQueryingEnumerable`1.AsyncEnumerator.InitializeReaderAsync(AsyncEnumerator enumerator, CancellationToken cancellationToken)
2024-10-16T21:06:09.479260760Z    at Microsoft.EntityFrameworkCore.Query.Internal.SingleQueryingEnumerable`1.AsyncEnumerator.MoveNextAsync()
2024-10-16T21:06:09.479263295Z Microsoft.Data.Sqlite.SqliteException (0x80004005): SQLite Error 1: 'no such column: c.RateLimitedUntil'.
2024-10-16T21:06:09.479265830Z    at Microsoft.Data.Sqlite.SqliteException.ThrowExceptionForRC(Int32 rc, sqlite3 db)
2024-10-16T21:06:09.479268355Z    at Microsoft.Data.Sqlite.SqliteCommand.PrepareAndEnumerateStatements()+MoveNext()
2024-10-16T21:06:09.479270829Z    at Microsoft.Data.Sqlite.SqliteCommand.GetStatements()+MoveNext()
2024-10-16T21:06:09.479273534Z    at Microsoft.Data.Sqlite.SqliteDataReader.NextResult()
2024-10-16T21:06:09.479276460Z    at Microsoft.Data.Sqlite.SqliteCommand.ExecuteReader(CommandBehavior behavior)
2024-10-16T21:06:09.479279045Z    at Microsoft.Data.Sqlite.SqliteCommand.ExecuteReaderAsync(CommandBehavior behavior, CancellationToken cancellationToken)
2024-10-16T21:06:09.479281549Z    at Microsoft.Data.Sqlite.SqliteCommand.ExecuteDbDataReaderAsync(CommandBehavior behavior, CancellationToken cancellationToken)
2024-10-16T21:06:09.479284084Z    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteReaderAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
2024-10-16T21:06:09.479286539Z    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteReaderAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
2024-10-16T21:06:09.479289003Z    at Microsoft.EntityFrameworkCore.Query.Internal.SingleQueryingEnumerable`1.AsyncEnumerator.InitializeReaderAsync(AsyncEnumerator enumerator, CancellationToken cancellationToken)
2024-10-16T21:06:09.479291598Z    at Microsoft.EntityFrameworkCore.Query.Internal.SingleQueryingEnumerable`1.AsyncEnumerator.MoveNextAsync()
2024-10-16T21:06:09.481636928Z [16-Oct-2024 23:06:09.479 FTL Program] Crashed on startup
2024-10-16T21:06:09.481644593Z Microsoft.Data.Sqlite.SqliteException (0x80004005): SQLite Error 1: 'no such column: c.RateLimitedUntil'.
2024-10-16T21:06:09.481647298Z    at Microsoft.Data.Sqlite.SqliteException.ThrowExceptionForRC(Int32 rc, sqlite3 db)
2024-10-16T21:06:09.481649853Z    at Microsoft.Data.Sqlite.SqliteCommand.PrepareAndEnumerateStatements()+MoveNext()
2024-10-16T21:06:09.481652357Z    at Microsoft.Data.Sqlite.SqliteCommand.GetStatements()+MoveNext()
2024-10-16T21:06:09.481654862Z    at Microsoft.Data.Sqlite.SqliteDataReader.NextResult()
2024-10-16T21:06:09.481657357Z    at Microsoft.Data.Sqlite.SqliteCommand.ExecuteReader(CommandBehavior behavior)
2024-10-16T21:06:09.481659881Z    at Microsoft.Data.Sqlite.SqliteCommand.ExecuteReaderAsync(CommandBehavior behavior, CancellationToken cancellationToken)
2024-10-16T21:06:09.481662496Z    at Microsoft.Data.Sqlite.SqliteCommand.ExecuteDbDataReaderAsync(CommandBehavior behavior, CancellationToken cancellationToken)
2024-10-16T21:06:09.481672845Z    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteReaderAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
2024-10-16T21:06:09.481675540Z    at Microsoft.EntityFrameworkCore.Storage.RelationalCommand.ExecuteReaderAsync(RelationalCommandParameterObject parameterObject, CancellationToken cancellationToken)
2024-10-16T21:06:09.481678105Z    at Microsoft.EntityFrameworkCore.Query.Internal.SingleQueryingEnumerable`1.AsyncEnumerator.InitializeReaderAsync(AsyncEnumerator enumerator, CancellationToken cancellationToken)
2024-10-16T21:06:09.481680700Z    at Microsoft.EntityFrameworkCore.Query.Internal.SingleQueryingEnumerable`1.AsyncEnumerator.MoveNextAsync()
2024-10-16T21:06:09.481683245Z    at Microsoft.EntityFrameworkCore.EntityFrameworkQueryableExtensions.ToListAsync[TSource](IQueryable`1 source, CancellationToken cancellationToken)
2024-10-16T21:06:09.481685920Z    at Microsoft.EntityFrameworkCore.EntityFrameworkQueryableExtensions.ToListAsync[TSource](IQueryable`1 source, CancellationToken cancellationToken)
2024-10-16T21:06:09.481688435Z    at TeslaSolarCharger.Services.Services.CarConfigurationService.AddAllMissingTeslaMateCars() in /src/TeslaSolarCharger.Services/Services/CarConfigurationService.cs:line 21
[... cut ...] onConvert.DeserializeObject[List`1](String value) at TeslaSolarCharger.Client.Helper.HttpClientHelper.<SendRequestWithSnackbarInternalAsync>d__8`1[[System.Collections.Generic.List`1[[TeslaSolarCharger.Shared.Dtos.LoggedError.DtoLoggedError, TeslaSolarCharger.Shared, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null]], System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext()`

Both may be unrelated to another. if i go back to :edge all works again

pkuehnel, 2024-10-16:
Which version number is displayed at the bottom of the Start page of TSC? Very likely you did not pull the latest `:latest`. `:edge` and `:latest` are currently the same.

b1ggi, 2024-10-16:
you were right, portainer stack did not pull new latest

[image]

now its good

closed 2024-10-16 by b1ggi