pkuehnel/TeslaSolarCharger #1573: Problems with Nginx Proxy Manager since V2.31.0

Toskache, 2024-10-22:
Hi!

First of all, thank you very much for the great project! Really great!
I access the TeslaSolarCharger via the Nginx Proxy Manager. This has worked great so far. However, with the current version (2.31.0) this no longer works. The left menu is still loaded, but the spinning circle in the middle keeps spinning.
I am not sure whether this is a bug or an incorrect configuration. I have attached the console log of the browser.

Best regards
George

[tsc.log](https://github.com/user-attachments/files/17472903/tsc.log)

pkuehnel, 2024-10-24:
Can you access the TSC directly? Can you check the network tab in your browser dev tools which connections fail? Very likely, TSC cannot upgrade the database structure. Can you restore an old version of the database?

Toskache, 2024-10-24:
Works fine now.

**Background**:
TSC was running on an RPI4 - also via an Nginx proxy manager. I then moved the TSC to a Proxmox LXC container. Of course, the IP changed in the process. Teslamate, Grafana and TSC worked with the new IP. Only TSC could not cope with the proxy and only ran via IP.
I have now switched off the old RPI4 and given the Proxmox LXC container the “old” RPI4 IP. And lo and behold... TSC works again behind the proxy.

Could it be that the “old” IP is permanently entered somewhere in the TSC config?

pkuehnel, 2024-10-24:
The only locations where you enter an IP/hostname are in the BaseConfigurationRazor (TeslaMateDatabaseHost, Mosquito servername, REST/Modubs/MQTT sources for solar values). But the error in your console log says there was something wrong when pulling the solar values. Very likely a caching issue: Frontend tried to deserialize a JSON which has another format now or called a REST endpoint which does not exist anymore. This should be fixed with the current version as Frontend autoreloads when a new server version is detected.

closed 2024-10-24 by pkuehnel