pkuehnel/TeslaSolarCharger #1592: GET /api/BaseConfiguration/UpdatePowerBuffer

bjv, 2024-10-31:
Hallo,

I know the TSC API is not official, but I found an API Endpoint in Swagger which is exactly what i was looking for:
GET /api/BaseConfiguration/UpdatePowerBuffer
How does it work?
GETting /api/BaseConfiguration/UpdatePowerBuffer?powerBuffer=250 does not seem to have an effect on the actual Power Buffer setting in the actual base configuration.
Is it not yet implemented completely and maybe a sneek peak on [Add editable powerbuffer to home page #1589](https://github.com/pkuehnel/TeslaSolarCharger/issues/1589)?

pkuehnel, 2024-10-31:
The base configuration shows the actually saved power buffer. The value you change with `/api/BaseConfiguration/UpdatePowerBuffer?powerBuffer=250` is only saved in memory, and after a restart of TSC it falls back to the configured value in Base Configuration. But when you set the value via `/api/BaseConfiguration/UpdatePowerBuffer?powerBuffer=250` that already works.

pkuehnel, 2024-10-31:
Note: currently there is a bug where if you only have the inverter power available the powerBuffer is multiplied by 2. So if you want to have 500Watt power buffer, you need to set 250Watt. Is resolved in the next version: https://github.com/pkuehnel/TeslaSolarCharger/issues/1593

bjv, 2024-10-31:
Ah, perfectly answered. All clear. Thank you!
That works even better for my purpose. 

Background explanation (for maybe a future feature): I want to handle my power buffer dynamically.
E.g. when being low on solar power (like 1kW incoming), i accept to get some power from grid to at least charge a bit.
Being on high solar power (like e.g. 7kW), i do not accept charging from grid, and maybe even want some positive buffer for temporary cloud coverings or stuff.
Just as an idea for a feature.
Currently I will handle that via the API and Home Assistant.

closed 2024-10-31 by bjv