pkuehnel/TeslaSolarCharger #1632: BLE command timeout

MrBetatester, 2024-11-17:
BLE stopped working after some time.

I paired both of our cars, but after some time I always get the error:

Sending command to tesla via BLE did not succeed. Fleet API URL would be: command/set_charging_amps. BLE Response: The request was canceled due to the configured HttpClient.Timeout of 22 seconds elapsing.

The BLE container is running on the same Pi. After pairing the cars, it seems to work, but after some days we get the timeout.

How can I check the container, or do we have to pair the cars again after some days?

Thank you

pkuehnel, 2024-11-17:
Which Pi do you use? How far is it away from the car(s). I also have two cars and am using TSC in two different locations. I had no BLE issues since I introduced it and never repaired anything.

MrBetatester, 2024-11-18:
I use a Raspberry Pi 4B. It is located in the middle between the two cars on the garage ceiling. The distance to each car is about 50cm.

pkuehnel, 2024-11-18:
Are you on the latest BLE Container? Tesla changed some things regarding reliability a few weeks ago.

MrBetatester, 2024-11-19:
I tried to update the Docker images today, but the TeslaSolarChargerBleApi was already up to date:

Pulling teslamate         ... done
Pulling database          ... done
Pulling grafana           ... done
Pulling mosquitto         ... done
Pulling teslamateapi      ... done
Pulling teslasolarcharger ... done
Pulling bleapi            ... done
TeslaSolarChargerBleApi is up-to-date
teslamate_grafana_1 is up-to-date
Recreating teslamate_database_1 ... 
Recreating teslamate_database_1  ... done
Recreating teslamate_mosquitto_1 ... done
teslamate_teslamateapi_1 is up-to-date
Recreating teslamate_teslasolarcharger_1 ... done

pkuehnel, 2024-11-19:
Please upload a log of your BLE container to https://nextcloud.th-deg.de/s/5zwqGRTzwBecxm9

Best would be if you had a timestamp when you repaired the car(s) and when it did not work. If you do not have a timestamp, please also add a Backup of your TSC because the errors are logged including timestamps.

MrBetatester, 2024-11-19:
I uploaded the log files.

pkuehnel, 2024-11-20:
I cannot see anything specific, it is always the same error: The car does not respond properly in time. The error is coming directly from Teslas tool to access the car, so I cannot do much about it. A new version has been released by Tesla a few days ago. I am currently testing it and will come back to you in a few days. If you hear nothing from me until Sunday, please comment with a brief reminder.

pkuehnel, 2024-11-24:
@MrBetatester Update the BLE Container to the latest image but instead of `:latest` use `:edge` as a tag. Does not change anything from my side but uses the latest software from Tesla.

MrBetatester, 2024-11-25:
I switched to Edge. Our Teslas also received version 2024.44.3.1 8c0c8e607685 during the night. After restarting the Pi, it has been running without any problems on the Y. Unfortunately, it still doesn't work on the Model3. I also noticed that the MY with the values from the M3 is displayed on the overview page. The M3 is not currently being managed because it always charges to 100% and there is no longer enough sun for both.

pkuehnel, 2024-11-25:
That is strange.. Are the VINs in the car settings page of TSC correct? I think the issue is, that somehow the cars are messed up, That might also be the reason why it does not work, because TSC tries to set values vor the Y but the Y is not at home resulting in the errors. Only if both cars are at home and plugged in, it works. Might that be the case?

pkuehnel, 2024-11-25:
Can you upload a Backup of your TSC in the link I provided before? Please also add a Text file where you write which car name has which VIN and add your E-Mail Address to the text file. I then will fix the database issues and send you a link via mail where you can download the fixed backup and restore it in your TSC.

MrBetatester, 2024-11-26:
I uploaded the files. This morning I also tried BLE again. 'Use BLE' and 'Use BLE for wake up' must be activated, then it works regardless of whether 'Should Be Managed' is activated. It also works regardless of whether the car is plugged in or not. I would recommend hiding the entire BLE section if 'Use BLE' is not activated. (Every now and then it also showed that it didn't work, but 10 seconds later it worked)

The swapped data is still the case.

[image]
[image]