pkuehnel/TeslaSolarCharger #1642: Spot Price + PV - car starts charging @ higher prices

gjuecat, 2024-11-20:
I pluged in the car and some time later I wondered, why the charge planning is divided into 2 parts, the first starts at 21:00 at higher prices for  17min and the second during the night at lower prices which looks OK.

Whats the purpose for the 17min charge after plugging in the car?
[image]
[image]

thx and best regards

pkuehnel, 2024-11-20:
Are you in Germany?

gjuecat, 2024-11-20:
Austria

pkuehnel, 2024-11-20:
Add the following Environment variable to you docker-compose.yml:

```
- AwattarBaseUrl=https://api.awattar.at/v1/marketdata
```

Note: You need to execute `docker compose up -d` so TSC does use the environment variable. Also note: It can take up to 48 hours until the changes take effect.

pkuehnel, 2024-11-20:
So, basically, the TSC started charging the car because the prices were low in Germany.

gjuecat, 2024-11-21:
thanks for the hint

closed 2026-10-17 by pkuehnel