pkuehnel/TeslaSolarCharger #1643: unhandled error

gjuecat, 2024-11-20:
since ~3 days I get this error on my PC, on the mobile phone everything is OK
any ideas?

[image]

thx and best regards

pkuehnel, 2024-11-20:
Which browser are you using, which operating system are you using? If you are on chrome please press F12 after the error occures. You should see a white X with a red background in the top right corner, click on it and send a screenshot of the error details (displayed at the bottom in red background).

pkuehnel, 2024-11-21:
Did you disable WebAssembly support? How do disable it is described here: https://github.com/stevespringett/disable-webassembly

WebAssembly needs to be enabled for TSC to work.

gjuecat, 2024-11-23:
I didn't change anything on both PCs. Chrome is started as everytime without an argument. I just tried with Edge and get the same error.
In Chrome it looks like it is on:
[image]

gjuecat, 2024-11-24:
> Chrome syncs all settings over all devices, so if you change the setting
> on one PC it won't work anymore on the other one as well. What happens if
> you use Firefox? (Edge also uses the Chromium engine). Also try a new
> Windows User account, start chrome (without login into Chrome) and try it
> there. If you do not get it running: Can you provide me with access to your
> TSC via VPN? As the only error message says that your browser does not
> support Web Assembly, I can not do much about it without direct access.

On another Win 11 PC with the same Chrome account everythings works fine,
also on a mobile phone with my Chrome account.
On the other Win 11 PC I added a new account on Windows, used the installed
Chrome but without Chrome Logon and I also get the unhandled error.

Would a Teamviewer or Anydesk Session be OK for you?

pkuehnel, 2024-11-25:
Ia m afraid I can't help you then. I thought maybe the issue could be reproduced with my chrome where I can use a debugger. But if it is working on another win 11 PC, also with Chrome I very likely cannot reproduce the issue with my Chrome (TeamViewer or AnyDesk would not help, as I need my dev environment). Try reinstalling Chrome and delete all chrome related stuff. (Not sure if this is up do date: https://support.google.com/chrome/thread/29816481/how-do-i-fully-uninstall-chrome?hl=en)

pkuehnel, 2024-11-25:
Please provide feedback if and how you could resolve the issue.

gjuecat, 2024-11-27:
I don't know why but now it works again on both (Win 10 and Win 11 PC) wirhout doing anything. Maybe a Chrome Update?? Windows Update isn't possible because they are not the same on Win 10 and Win 11.

closed 2024-11-27 by gjuecat