pkuehnel/TeslaSolarCharger #1644: Feature Request: Lower Minimum Car Power Adjustment Interval

rsaul93, 2024-11-21:
### Use Case
For setups with low-latency connections (e.g., local Modbus TCP and BLE API), users would benefit from the ability to set the Car Power Adjustment Interval more freely, potentially as low as 5 seconds. The current minimum of 25 seconds can delay responses to rapidly changing solar conditions, such as on partly cloudy days, reducing efficiency in utilizing solar surplus.

### Proposed Feature
Allow users to configure the Car Power Adjustment Interval without the current minimum threshold of 25 seconds. Include:

**Default Setting:** Retain the default interval of 30 seconds for stability.
**User Warning:** Display a warning for intervals below 25 seconds, e.g.:
_"Short intervals may cause system instability. Recommended only for setups with low-latency connections. Use with caution."_

### Additional Proposals to Prevent Oscillations (mitigate potential instability from shorter intervals):

**Adaptive Minimum Interval:** Enforce a dynamic minimum interval based on measured car response lag (e.g., using inverter data to infer response time).
**Anticipate the Car’s Power Draw:** Infer the car's current power draw indirectly, as BLE API does not provide feedback. Use inverter or energy meter data to estimate the draw based on the last issued command and elapsed time. Continuously refine this estimate by monitoring the system’s response and stabilizing patterns after power adjustments.
**Floating Power Adjustment Interval:** Replace fixed intervals with a dynamic, need-based approach. Measure solar generation and estimated car power draw at high frequency (e.g., every second) and intervene only when deviations exceed a defined threshold. 

### Benefits

**Flexibility:** Freely configurable intervals suit advanced setups.
**Efficiency:** Improved response to rapid solar power changes.
**Stability:** Safeguards like adaptive intervals prevent oscillations.
This balances advanced user needs with system reliability while maintaining robust default behavior for general users.

rsaul93, 2024-11-25:
I observed an error yesterday while charging for several hours:

"Error with key CarRateLimited for car LRW*** in TeslaFleetApiService.SendCommandToTeslaApi: Car is rate limited until 11/25/2024 00:00:00"

I'm puzzled as to why this error was thrown at all since the car is connected via BLE. The "API refresh interval" was set to 30 and was the same on a day when no charging was ongoing.

Shortly before the charging was finished, these errors started to occur, leading to TSC not retrieving the current amperage the car was charging with. This resulted in the following scenario:

Charging was stopped via the charge limit set in the Tesla app.
TSC still gathered the correct solar generation and grid power data.
However, it continued to anticipate the car to be charging at the last amperage before the error occurred.
This led to a negative home consumption reading, even though the four components (solar generation, grid power, house consumption, car consumption) can be inferred directly from data that was still available to TSC, despite the Tesla cloud API being rate-limited.
I've attached screenshots from the power draw overview (which shows a negative house consumption—even though I don't have a home battery configured or installed) and the last charging stats (which show a significantly higher used solar energy than expected).
[image]
[image]

pkuehnel, 2024-11-25:
You need to set the API refresh interval to ~500 seconds. The Fleet API is also rate limited for getting data from the car, not only sending it. So setting the refresh intervall to 30 seconds will result in the rate limit you were experiencing. The reason why there was no rate limit yesterday: The check if the car is awake or asleep is not rate limited, so I only request the car if it is awake. There are also many other optimizations to prevent rate limits, e.g. if your car is not at home, the earliest home arrival time is calculated (simulated drive with 100 kph to home in direct line) and the car gets no request until that time. After TSC sends a command to the car, it is always requested to detect if the command was successful. The only issue you will experience when setting the API refresh interval to 500 seconds: TSC needs up to 500 seconds until it detects that the car is at home and plugged in. These two issues will be resolved in the next release of TSC by using Fleet telemetry (needs to be enabled in the car settings of TSC).

pkuehnel, 2024-11-25:
And additionally: As the car was rate limited, TSC could not detect any changes in charging speed, soc,... This also leads to the wrong values in the charging process, as TSC thought, the car is still consuming 5kW and only uses 100 Watt from the grid, so the rest has to come from solar, resulting in used solar energy being 79 kWh for the charge. The solar power (which is 0 Watt in your screenshot) is not used for anything but showing the value in the UI. Everything is only calculated by using the grid power.

pkuehnel, 2024-11-25:
@rsaul93 This is off topic here, Check out my answers here: https://github.com/pkuehnel/TeslaSolarCharger/issues/1650

closed 2025-06-10 by pkuehnel