pkuehnel/TeslaSolarCharger #1661: another option for ChargeMode

bff0172, 2024-11-26:
When the weather is cloudy and very variable I want to setup a fixed ampere below 5A (Tesla app only can set 5) because I don't have a house battery.
Until now I set this fixed ampere (like 2) with BLE going in container and do:
/app/go/tesla-control, -ble -vin xxx -key-file /externalFiles/privateKey.pem charging-set-amps 2

It would be good to have this option in selection list - for example: set fixed charging to minimum amperage

pkuehnel, 2024-11-26:
For now you can set the maximum current in the car settings to 2A, then set the charge mode to max power, TSC should set the current to 2A then.

pkuehnel, 2025-06-10:
Will be possible in the next version