pkuehnel/TeslaSolarCharger #1663: New issues after v2.31.0

raghu1121, 2024-11-27:
I see the following issues after v2.31.0, is it me or someone else also has these ?

- I reach home in the evening with about 10-15%, when i connect to wallbox the car starts charging immediately, but it should stop after a minute or two in PV only mode. It doesn't stop, so i stop it manually and some time later it starts to charge again. It charges even when there is no PV at which ever power level was set in the App.
- The Power buffer acts weirdly. It used work normally, until recently, although i put to X watts it considers (x+600) watts and draws additional power, but sometimes works normally again.
- Sometimes power buffer doesn't get updated in the base configuration when I update at the overview page, I have to manually check if it is updated at both places.
- On a cloudy day, when there is enough power car starts charging with PV only, then when there are clouds and all covered it doesn't reduce the power it keeps going. I have to manually stop for it to adjust the power. for example When ever there is huge amount of PV power, it goes to peak charge power of 11kW, then when clouds covers, it doesn't reduce, it keeps going at 11kW for some time.
- I think when ever there is a huge change in the power it's better to stop and start charging, which happens fast rather than to adjust values which takes forever to adjust. Is there a possibility to do that? as my consumption is actually high on these windy and fast moving cloudy days as the power adjustment is slow.
- Sometimes the charge doesn't start, even after getting enough power for about few minutes i have to manually start and stop , then it starts and then adjusts the power. 
- I am using BLE, what is the polling interval for this? Is it automatically set? I only see API refresh interval in settings.

If you have solutions to my problems, please share your suggestions. Please also let me know what info do you need to look into these issues in more detail.

pkuehnel, 2024-11-29:
Already answered here: https://tff-forum.de/t/teslasolarcharger-pv-ueberschussladen-mit-beliebiger-wallbox-teil-2/350867/844

pkuehnel, 2024-11-30:
Most of your issues should be resolved in the latest version, please update and try again. If the issues persist, please come back here.

closed 2024-11-30 by pkuehnel