pkuehnel/TeslaSolarCharger #1718: Wrong "No Fleet API requests allowed" Error

RicoKraft, 2025-01-07:
Hey there,
app is working fine from what I can tell.  TSC is communicating with my Tesla and starts/stops charges. Token has been reset to make sure, that theres no issue. TSC is on the latest version v2.34.0, Internet access is there, but the error still persists.

The issue started to appear since updating from 2.31 to 2.34.

Best regards, Rico

pkuehnel, 2025-01-07:
Please restart the TSC container and about 1 minute after restart [create a logfile](https://github.com/pkuehnel/TeslaSolarCharger?tab=readme-ov-file#generate-logfiles) and upload it [here](https://nextcloud.th-deg.de/s/fka8tzCiSRAFzab).

RicoKraft, 2025-01-07:
Done! In addition there's a new error message, which disappears after resetting/generating the fleet api token again.

[image]

pkuehnel, 2025-01-07:
Will look into it within the next 20 hours.

pkuehnel, 2025-01-08:
@RicoKraft The logs are not detailed enough from my side. I just added a few log statements. Could you please, change the image in your docker `docker-compose.yml` from `pkuehnel/teslasolarcharger:latest` to `pkuehnel/teslasolarcharger:fleetApiAllowedLogging`. Then run `docker compose pull`, then `docker compose up -d` and after about a minute again upload the log files to the link I posted yesterday?
Thanks in advance.

RicoKraft, 2025-01-10:
@pkuehnel Done. Sorry for the delay, was out of town. Same again. Had the issue regarding the vehicle information again, reset the token again and then created the logfile. New logfile is called "teslasolarcharger (3).log" Another information, that might be important: I'm running this on a Synology. Thanks for your help!

RicoKraft, 2025-01-10:
Oh and on car settings theres this error aswell:
[image]
All issues seem to indicate that theres no token at all and TSC doesnt have access to the car. I dont understand why its still able to start/stop charges though and TSC is still working as intended.

pkuehnel, 2025-01-10:
Looks like you (at least TSC on your Synology NAS) can not open `https://www.teslasolarcharger.de/api/Tsc/AllowUnlimitedFleetApiAccess?installationId=asdf` within 2 seconds. Any Firewall, DNS Filters or something like that active? You can test the link in the browser as well.

RicoKraft, 2025-01-10:
If I open it on my browser it says "{"value":true}" immediately. I'm not sure how I can test that on Synology right now. I don't have any Firewall or DNS Filters active as far as I know or I can tell.

pkuehnel, 2025-01-11:
Strange. I increased the timeout from 2 to 100 seconds and further improved the logging. Please let me know if it works now. No matter if it works again, upload the log, so I can see how many seconds it takes until you get a response. To update to the latest version, just do another `docker compose pull` and `docker compose up -d` still with the image `pkuehnel/teslasolarcharger:fleetApiAllowedLogging`

RicoKraft, 2025-01-11:
@pkuehnel Done. I think we got it! Error disappeared, new log uploaded, it's number 4. Let me know if you need anything more. I'll keep you updated if any Error comes up later.

pkuehnel, 2025-01-13:
Still very strange why that takes so long. It should take 100ms max 🤔

I close this issue as this won't be needed anymore in the next TSC version. Note: As Tesla starts charging for Fleet API in February, I will flip the switch hopefully in about two weeks, so the error will come up again for you, but then it's not a bug but a feature and you need to update to the latest `:latest` version then.

closed 2025-01-13 by pkuehnel

pkuehnel, 2025-01-14:
When you see the next release, you can go back to latest.