pkuehnel/TeslaSolarCharger #1719: ModBus configuration details in Readme unclear

sftman18, 2025-01-07:
Reading the documentation, I was led to believe that ModBus required a plugin configuration in Docker, however I was unable to find the details on what that should be.
Further in the Readme, there are some details about ModBus with the title "Modbus values" however the details aren't tied into "where" they are configured.

I managed to figure out that there was an undocumented "Add new Modbus TCP source" under the Base Configuration next to REST sources.

I think it would be more clear, if after the REST API examples in the Readme, there was one showing how Modbus TCP may be configured.

Thanks for your efforts!  I'm looking forward to using it!

pkuehnel, 2025-01-07:
Thanks for the info, in previous versions there was a dedicated Modbus Plugin, looks like I did not fully update the docs. Will be fixed.

sftman18, 2025-01-07:
Great, just trying to help make things clearer.