pkuehnel/TeslaSolarCharger #1722: Solaredge API Plugin - Grid Power Always Zero

pushimi, 2025-01-11:
Description:
The Solaredge API plugin isn’t working as expected. The gridPower value always shows as 0, no matter what the actual grid usage is. Also power buffer doesnt work.

I also tried adding http://teslasolarcharger/api/Index/GetPvValues as the source for gridPower, but it doesn’t seem to work either.

Questions:

1. Is there a way to debug or troubleshoot why the gridPower value is incorrect?

2. Are there specific requirements for the API endpoint format or data?

Let me know how I can fix or debug this issue.

![Screenshot_20250111_131628_Chrome.png](https://github.com/user-attachments/assets/c0c67361-2030-41fd-8f80-87f4[image]

.png …]()

[image]

[image]

[image]

pkuehnel, 2025-01-11:
This is expected behavior, check out https://github.com/pkuehnel/TeslaSolarCharger?tab=readme-ov-file#content-using-solaredge-plugin

pkuehnel, 2025-01-11:
Note after every restart, the Plugin "forgets" that it is in the rate limit, tries again and again for 15 minutes, the API is rate limited. Also the test button and even opening the settings forces a call and the 15 minutes start again. Just wait 15 minutes without doing anything and the Plugin should recover. If after 15 minutes it stoll shows 0, let me know.

pushimi, 2025-01-11:
Yes its still zero. I followed your instructions for setting up the plugin but even when I am producing solar AND importing from grid, the grid shows zero, but solar is correct value. What can I do next?

pkuehnel, 2025-01-11:
That is because only Grid power and home battery power are relevant for charging purposes, that's why only grid power and home battery power are faked to 0 (like described in the README). The only thing I could imagine is that SolarEdge changed their rate limits (as in their documents they state that the 300 requests are per day and the 15-minute waiting is just a workaround I found a few years ago).

If you still think something is wrong, please upload the logs of the solarEdge plugin [here](https://nextcloud.th-deg.de/s/t3gBwGAHmWAQGHd)

[How to generate logfiles](https://github.com/pkuehnel/TeslaSolarCharger?tab=readme-ov-file#generate-logfiles)

pushimi, 2025-01-11:
Yes Sir, will do

pkuehnel, 2025-01-11:
Please let me know when it is online, as I do not get a notification on new files.

pushimi, 2025-01-12:
I have just uploaded both the SE log and TSC log (its 1gig large lol).

Please lmk when if you figure anything out. I am available for any further questions.

pkuehnel, 2025-01-12:
I checked it out, I can not find an error, but what I found is, that the SolarEdge Plugin always uses 0 as value, even if a Power Buffer (in your case 3500) is set, which results in charging behavior not being as expected.

On additional note: only if at least one car is `relevant` which means:
1. Car is at home
2. Car should be managed is enabled in Car settings
3. Charge Mode i no `TSC Disabled`
4. The car is plugged in
5. Climate is on or car is pulling at least 1A or state of charge at least 3% lower than than the SoC Limit.

The last time all these conditions were met in your installation was Today at 14:42. And at that point TSC displayed 5610 Watt feeding into the grid.

My recommendation: Set your Power Buffer to 0 as otherwise TSC won't work properly. I will fix the power buffer related issue (but due to changes in Tesla's API which are more important to fix currently, it will take some time).
Another option for you would be to try to get the relevant values directly from you inverter via Modbus/local REST API,... but not sure how this works with SolarEdge, maybe [this](helps).

pushimi, 2025-01-13:
I will wait until we get some sun and then i will see. Do i need a specific modbus plugin installed?

pkuehnel, 2025-01-13:
Ok.
No, not needed.