pkuehnel/TeslaSolarCharger #1778: TSC not working anymore after upgrading to v2.35.0 and Solar4Car Backend

owkowo69, 2025-01-25:
Hi,

TSC was running fine until the latest upgrade (2.35.0).

After the upgrade and migrating to Solar4Car (Base + Fleet Api subscription), SOC is not displayed correctly and charging is not started in "Maximum Power" or "Sport Price + PV" Modes.
Cloud Connection shows "Solar4Car Backend connection" and "Tesla Fleet API Connection" both green.

I found errormessage

"2025-01-25T23:21:04.211894239Z [26-Jan-2025 00:21:04.210 ERR TeslaSolarCharger.Server.Services.BackendApiService] Error while sending request to backend

2025-01-25T23:21:04.214593885Z System.Text.Json.JsonException: The input does not contain any JSON tokens. Expected the input to start with a valid JSON token, when isFinalBlock is true. Path: $ | LineNumber: 0 | BytePositionInLine: 0."

in Logfiles, see attachment.

Thanks for your support,
Wolfgang

[_teslasolarcharger_logs.txt](https://github.com/user-attachments/files/18548337/_teslasolarcharger_logs.txt)

pkuehnel, 2025-01-26:
Please upload logs (not created with Portainer, but like described [here](https://github.com/pkuehnel/TeslaSolarCharger?tab=readme-ov-file#generate-logfiles))
To: https://nextcloud.th-deg.de/s/sgoG9HWspkqS3k8

Please also upload a Textfile where you insert your VIN, InstallationID and the email address you used to register an account on Solar4Car.com

pkuehnel, 2025-01-26:
I resolved a bug about 5 hours ago, which could have lead to your issue. Please again check if it is working properly now.

pkuehnel, 2025-01-26:
On which hardware is your TSC running, there is at least something wrong with the encoding of your log file, it is impossible for me to read it.

pkuehnel, 2025-01-26:
You are using Fleet API with a Model Y or Model 3, correct?

If you can't fix the encoding, remove the verbose log again from your `docker-compose.yml` , update to the latest version again, set the car to max Power while it is not charging and send me a log using Portainer, but make sure to do the following:

1. Set lines to 10000 (from my experience this is about the maximum without the browser page crashing
2. Wait for the browser to reload the logs, so you can see a tiny scrollbar on the bottom right
3. Disable Auto-refresh logs (otherwise the page crashes)
4. Download the logs

[image]
These logs then please upload to my link again.

owkowo69, 2025-01-26:
Hi,

using 2.35.2 it's working again, switching from "max power" to "spot price" multiple times was fine.
The logfile was "ansi color encoded" by dockers console option -tty, i uploaded the file again if you want to check it anyway.

Thanks a lot for your work and fast response.

Kind regards,
Wolfgang

pkuehnel, 2025-01-26:
Ah okay. Thanks, I will check it anyway as I did not change anything that would resolve an issue like that.

pkuehnel, 2025-01-26:
Issue was that car was at 98%, TSC does not start if Soc Limit is not at least 3% higher than actual Soc because sporadically the cars do not start charging if to close to the limit even though it is below. That would result in the car never going to sleep and TSC trying to start charging forever. Can you confirm that this might have been the case?

owkowo69, 2025-01-26:
No, because at the time of my first post SOC was about 25%.
I used Tesla scheduling to charge this night, thats the reason for 98% today.

pkuehnel, 2025-01-26:
Do you have a timestamp of the issue from the day before?

owkowo69, 2025-01-26:
Unfortunately not a exact one, yesterday between 22 - 23:50.

pkuehnel, 2025-01-26:
Logs are starting at 00:18 sadly. I will close this now, please reopen if the issue comes back.

closed 2025-01-26 by pkuehnel