pkuehnel/TeslaSolarCharger #1802: Kostal Plenticore Grid Power not working anymore

subjacked, 2025-02-01:
I don't know when this happened exactly. I'm sure my configuration worked the whole time, now my grid power shows 0 and the home consumption is the sum of negative power grid and the actual home consumption...

[image]

[image]

pkuehnel, 2025-02-01:
I assume that you never fed power to the grid only consumed and Kostal has a different register for that. Could that be the case?

subjacked, 2025-02-01:
You see in my second screenshot that I fed 3.8 kw to the grid over a long period of time.

pkuehnel, 2025-02-01:
Did it work anytime in the past or are you a new user and this is the first time you send power to the grid? My assumption is, that the Kostal register 108 as you configured, never contains a negative value as it might be the register for "grid consumption" which is `0` when you feed power into the grid. There might be another register that is for `grid feed in` which is `0` when you consume power from the grid but shows a value wheen feeding power into the grid.

What I meant: As it was winter, you never had the case that power was feed into the grid before and you just did not notice that is has never worked.

pkuehnel, 2025-02-01:
I just checked out https://www.photovoltaikforum.com/core/attachment/81082-ba-kostal-interface-modbus-tcp-sunspec-pdf/

on page 14 in the PDF (9 is printed as page number on that page) register `108` says `Home own consumption FROM grid`.

pkuehnel, 2025-02-01:
The correct value might be 252 or 254 or 256 depending on the location of your sensor (I am not into any of that, just read the PDF).

https://github.com/pkuehnel/TeslaSolarCharger/discussions/174#discussioncomment-3283275

The user there mentions the register 252 as the correct register. Note: As 2022 the UI was completly different, therefore there are URLs posted, which are not needed anymore. As far as I see that, you just need to change the register address from 108 to 252 and it should work (still depending on the "Sensor position" but like I said, no idea what that means.

subjacked, 2025-02-01:
I am pretty sure it worked correctly. I use TSC for over a year now.

subjacked, 2025-02-01:
[image]

closed 2025-02-01 by subjacked

pkuehnel, 2025-02-01:
Hmm, strange, I am not aware of any change from my side there as I have been working pretty hard to fix everything regarding Tesla API cost for about three months and I skipped everything else that is not extremely important. And I am still not sure if it is really working now. It just shows a negative number now, so consuming power from the grid. Wait for tomorrow, when you again feed power into the grid.

subjacked, 2025-02-02:
[image]

It seems to really work again.