pkuehnel/TeslaSolarCharger #1803: Step 9. Restart TSC to add car not working - token problem?

JoKo1714, 2025-02-01:
After restarting TSC (as decribed in the README: First startup of the application at step9) the car is not listed  in the inventory and cannot be managed

Error log

[01-Feb-2025 22:26:08.220 ERR TeslaSolarCharger.Server.Services.TeslaFleetApiService] Error while getting all cars from account due to communication issue between Solar4Car Backend and Tesla: Underlaying Status code: OK; Underlaying Result: {"response":[{"id":929885853710367,"vehicle_id":1689303238173470,"vin":"XP7YGCES6RB×××××","color":null,"access_type":"OWNER","display_name":"HH-*****","option_codes":null,"granular_access":{"hide_private":false},"tokens":null,"state":"asleep","in_service":false,"id_s":"929885853710367","calendar_enabled":true,"api_version":84,"backseat_token":null,"backseat_token_updated_at":null,"ble_autopair_enrolled":false}],"pagination":{"previous":null,"next":null,"current":1,"per_page":1,"count":1,"pages":1},"count":1}m×××××@raspberrypi2:~/pTeslaSC $

JoKo1714, 2025-02-01:
[01-Feb-2025 22:04:48.459 DBG TeslaSolarCharger.Server.Services.ErrorHandlingService] 0 errors removed as did not reach minimum error count
[01-Feb-2025 22:11:28.052 DBG TeslaSolarCharger.Server.Services.ErrorHandlingService] 0 errors removed as did not reach minimum error count
[01-Feb-2025 22:11:31.519 DBG TeslaSolarCharger.Server.Services.ErrorHandlingService] 0 errors removed as did not reach minimum error count
[01-Feb-2025 22:14:44.507 DBG TeslaSolarCharger.Server.Services.ErrorHandlingService] 0 errors removed as did not reach minimum error count
[01-Feb-2025 22:16:33.103 DBG TeslaSolarCharger.Server.Services.ErrorHandlingService] 0 errors removed as did not reach minimum error count
[01-Feb-2025 22:16:35.409 DBG TeslaSolarCharger.Server.Services.ErrorHandlingService] 0 errors removed as did not reach minimum error count
[01-Feb-2025 22:24:29.680 DBG TeslaSolarCharger.Server.Services.ErrorHandlingService] 0 errors removed as did not reach minimum error count
[01-Feb-2025 22:26:06.266 INF TeslaSolarCharger.Server.Services.ErrorHandlingService] Delete 0 logged errors as they have unknown keys
[01-Feb-2025 22:26:23.583 ERR TeslaSolarCharger.Server.Services.BackendApiService] Could not load new Backend Information. https://api.solar4car.com/api/Tsc/GetBackendNotifications?installationId=6bb189be-415b-4ea0-9704-1bab0a2897a3&lastKnownNotificationId=0: Unexpected error: The input does not contain any JSON tokens. Expected the input to start with a valid JSON token, when isFinalBlock is true. Path: $ | LineNumber: 0 | BytePositionInLine: 0.
malte@raspberrypi2:~/pTeslaSC $

pkuehnel, 2025-02-01:
Is fixed in `2.35.5`.

pkuehnel, 2025-02-01:
Forgot to create a new version a few days ago.

closed 2025-02-01 by pkuehnel