pkuehnel/TeslaSolarCharger #1811: Feature Request: Support/Implement full local BLE

bjv, 2025-02-06:
Hi Peter,

referring to https://github.com/pkuehnel/TeslaSolarCharger/issues/1800, first thanks for your information.

Then, I would like to place the Feature Request to implement full local BLE support without any need of cloud telemetry or support.
These would keep costs low and also increase security since everything regarding the car is handled locally.

As from your words in #1800 I am not sure if you are aware, that it would be possible to keep TSC (including all relevant telemetry) completely local without any need of official telemetry, API, cloud or costs.

I (kind of) re-implemented TSC (my relevant features, "Überschussladen" and "MinSoC") locally with the help of https://github.com/tesla-local-control/tesla_ble_mqtt_docker and Home Assistant now. 
Since the mentioned project offers all BLE telemetry and commands via MQTT, it may also be easy to integrate into TSC.

As a POC with this Projekt and HA, I went completely local and - sad to say - as a former supporter moved away from TSC.
If the project would have been in another language I would have done my implementations as a PR for TSC.

Instead I maybe will have to wait if local BLE Support will be coming.

Anyways, thanks a lot for all of your work; I had a great time using TSC so far.

pkuehnel, 2025-02-06:
Is planned, currently the main issue is, that BLE is somewhat instable for some people (especially when trying to poll data when the car is away, so that's the main reason for Fleet Telemetry being implemented: Detection, if the car is at home), which is currently unknown why this happens, see

https://github.com/tesla-local-control/tesla_ble_mqtt_docker/issues/67
https://github.com/tesla-local-control/tesla_ble_mqtt_core/issues/142
The root cause is on Teslas underlaying BLE communication package, see
https://github.com/teslamotors/vehicle-command/issues/272
https://github.com/teslamotors/vehicle-command/pull/370
https://github.com/teslamotors/vehicle-command/pull/353

Currently some people are working on fixing that. As soon as this is fixed, I will implement that including a bunch of other things like getting Soc for other car brands and OCPP integration. As this will be a major rework it will take some time. I want to be sure, that BLE is as stable as possible as I would rather not spend much time on debugging and support instead of further developing TSC, so implementation will come as soon as BLE is super stable.

Just as a notice: I never had these issues, but some have. Your liked project has the same issues.

Lenart12, 2025-02-07:
Hi, I just want to offer my input since you mentioned 3 of my PRs in vehicle-command repo to add that I'm already using all the new features/fixes you talked about in my project https://github.com/Lenart12/TeslaBle2Mqtt which i specifically made because I was unhappy with how slow and unreliable tesla-local-control was. And so far it has been super stable (way more than tesla-local-control in my testing). Probably your best bet is to use TeslaBleHttpProxy and adapt it for full local BLE control as it is the least amount of work for you (first https://github.com/wimaha/TeslaBleHttpProxy/pull/95 needs to gets merged).

pkuehnel, 2025-02-07:
But you use the https://github.com/teslamotors/vehicle-command as the underlying code for communication with the car, right? Which parameters do you use to call it in a stable way? Can you give me the source doe from your project? (I have absolutely no idea of Go)

Lenart12, 2025-02-07:
That's correct. The main benefit of using TeslaBleHttpProxy which uses vehicle-command over tesla-local-command (tlc) is that it does not keep recreating (resetting) the BT adapter connection which causes issues many issues on linux ([see here](https://github.com/teslamotors/vehicle-command/blob/82b4899b2c904b0268d78d86eb557b7220b169f8/pkg/connector/ble/ble.go#L139)). In contrast tlc by design can not reuse the same device "connection" because the vehicle-control utility can not do so and so for every command resets the adapter which causes issues.

My code is here https://github.com/Lenart12/TeslaBleHttpProxy

pkuehnel, 2025-02-07:
> My code is here https://github.com/Lenart12/TeslaBleHttpProxy

I meant the specific part where you are calling Tesla's `tesla-control`

pkuehnel, 2025-02-07:
> In contrast tlc by design can not reuse the same device "connection" because the vehicle-control utility can not do so and so for every command resets the adapter which causes issues.

Does this mean TeslaBleHttpProxy does not support multiple VINS at the same time?

Lenart12, 2025-02-07:
It does, I mean BT "connection" as in socket to unrelying HCI device, which can then support multiple clients.

pkuehnel, 2025-02-07:
Thanks for your input, will look into it.

pkuehnel, 2025-03-05:
@Lenart12 I installed the BLE HTTP Proxy yesterday and tried a few things. The issue I have is, that it does not return the same results. E.G. If I start charging while the car is asleep your proxy autodetects that and wakes up the car. I do not mind that that much, but if charging of the car is completed (e.g. because the charge limit is reached), when I start charging via your http proxy, I do not get any error message. The return message is not different from a successful charge start. The error message is logged as an info log in your container, though. I found this pull request: https://github.com/wimaha/TeslaBleHttpProxy/pull/95
Will the issue I mentioned above be fixed with that PR?

Lenart12, 2025-03-07:
I think you need to add `?wait=true` to your POST requested, otherwise its just fire&forget. By adding that parameter it waits for completion. This is already working on the main branch. Regarding waking up the car, to avoid this you need to fetch body controller state and check the sleep status before deciding if sending command. This works on main but its slightly improved on my PR, however it is not stable enough to use in prod...

pkuehnel, 2025-03-07:
I am pretty sure I did. Can not double check currently as the car won't be at home for a few days.