pkuehnel/TeslaSolarCharger #1814: Error while getting vehicle info for car.... - Could not deserialize vehicle: null

gjuecat, 2025-02-07:
Hi,

since ~1 week TSC isn't working. I think this happend with the update to the paid version. I did a free trial with this version but I'm also using BLE and therefore my preferred solution would be the free BLE one.

Now I always get this errors:
[image]

BLE is working, did the tests with wake up and set 7A.

Any ideas?
Thank and best regards

pkuehnel, 2025-02-07:
A free version is not possible anymore, as Fleet Telemetry data is required, which costs me money.

pkuehnel, 2025-02-07:
Will add a correct message to clarify this.

pkuehnel, 2025-02-07:
Further Details: https://github.com/pkuehnel/TeslaSolarCharger/issues/1815

closed 2025-02-07 by pkuehnel