pkuehnel/TeslaSolarCharger #1830: Backup

giss69, 2025-02-11:
Hallo,

ich wollte auf neues TSC umstellen ohne Teslamate und habe mir einen Backup erstellt "2.35.6_TSC-Backup".
Ich kann aber diesen Backup nicht wieder einspielen, wenn ich die Datei auswähle passiert einfach nichts.
Woran könnte es liegen? Kann man irgendwie anders Backup aufspielen?

Danke

pkuehnel, 2025-02-11:
Nach Auswählen der Datei über "Select Backup File" sollte darunter die Datei incl. Größe angezeigt werden und dann kann man auf den Start restore Button klicken. Nach einem Klick darauf, sollte rechts oben eine Erfolgsmeldung kommen, danach ist ein Neustart des TSC erforderlich. An welcher Stelle passiert nichts?

[image]

giss69, 2025-02-11:
geht leider nicht, Datei kann ich zwar auswählen aber darunter wird sie ( incl. Größe) nicht angezeigt und Button bleibt ausgegraut.

pkuehnel, 2025-02-11:
Komisch, kannst du mal das Backup hier hochladen: https://nextcloud.th-deg.de/s/YRCC4WTWHejNbCP

pkuehnel, 2025-02-11:
Funktioniert bei mir ohne Probleme:

[image]

Ich nutze Chrome auf Windows, du?

giss69, 2025-02-11:
hmmm, dann liegt es evt. an Safari.
Ok. Danke!

closed 2025-02-16 by pkuehnel