pkuehnel/TeslaSolarCharger #1916: Max current ignored when charging to Min SOC?

BillGOH, 2025-04-05:
I have two Tesla Wall Connectors Gen2. There are on the same fuse with max current of 3x16A. Therefore TSC base configuration max current value was set to 16A. This works fine for solar charging. But when I set workday charging to Min SOC for both cars the value seems to be ignored. Both cars charge with up to 16A, therefore the fuse is triggered.

Best solution would be - in my opinion - that cars with the same priority share the max current; in my case 8A each.

Newest version of TSC (2.36.4) is used with Tesla Fleet API.

BillGOH, 2025-04-05:
Today during solar charging both cars were charged with more than 16A (11kW). Before, I had set the same priority for both cars. After resetting the fuse TSC did not react when both cars tried to charge with 16A. I had to reduce the charge current via Tesla App. Last year everything worked fine in this regard. 
[image]

pkuehnel, 2025-04-05:
What did you set in the BaseConfiguration as `Max combined current`?

BillGOH, 2025-04-05:
Thanks for your reply. I set 16A.

pkuehnel, 2025-04-05:
As I will reprogram the whole charging service it will take about 1-2 months to fix it.

BillGOH, 2025-04-06:
Thanks for your efforts! Of course I meant max combined current in my first post. Max current per car is 16A, as well as max combined current.